adafruit/Adafruit_FRAM_I2C #2: new read and write functionalities

gabrydiamonds, 2016-01-04:
First of all thanks for your development for the community.
I would offer to you and the community these new read and write functionalities:
this library writes any types of values at the specific FRAM address.
This code is used to write any data type (int, float, double, string, char, structures, etc.) to an I2C FRAM
In the normal write anything the framaddress is incrimented after the writing of each byte.
The Wire library does this behind the scenes.

I inspired from an eeprom_i2c library code found in the web here:
http://apexlogic.net/code-bank/arduino/eeprom-i2c-write-anything/

It would be "framstastic" if this code will be implemented on the master library Adafruit_FRAM_I2C.h

Here the header code:

/**************************************************************************/
/*! 
    @file     Adafruit_FRAM_I2C.h
    @author   KTOWN (Adafruit Industries)

    @section LICENSE

    Software License Agreement (BSD License)

    Copyright (c) 2013, Adafruit Industries
    All rights reserved.

    Redistribution and use in source and binary forms, with or without
    modification, are permitted provided that the following conditions are met:
    1. Redistributions of source code must retain the above copyright
    notice, this list of conditions and the following disclaimer.
    2. Redistributions in binary form must reproduce the above copyright
    notice, this list of conditions and the following disclaimer in the
    documentation and/or other materials provided with the distribution.
    3. Neither the name of the copyright holders nor the
    names of its contributors may be used to endorse or promote products
    derived from this software without specific prior written permission.

    THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS ''AS IS'' AND ANY
    EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
    WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
    DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER BE LIABLE FOR ANY
    DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
    (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
    LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
    ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
    (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
    SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
*/
/**************************************************************************/
#ifndef _ADAFRUIT_FRAM_I2C_H_
#define _ADAFRUIT_FRAM_I2C_H_

#if ARDUINO >= 100
 #include <Arduino.h>
#else
 #include <WProgram.h>
#endif

#include <Wire.h>

#define MB85RC_DEFAULT_ADDRESS        (0x50) /* 1010 + A2 + A1 + A0 = 0x50 default */
#define MB85RC_SLAVE_ID       (0xF8)

class Adafruit_FRAM_I2C {
 public:
  Adafruit_FRAM_I2C(void);
  
  boolean  begin(uint8_t addr = MB85RC_DEFAULT_ADDRESS);
  void     write8 (uint16_t framAddr, uint8_t value);
  uint8_t  read8  (uint16_t framAddr);
  void     getDeviceID(uint16_t *manufacturerID, uint16_t *productID);
  
  //added functions
  	template <class T> uint16_t write(uint16_t framAddr, const T& value);
  	template <class T> uint16_t read(uint16_t framAddr, T& value);

 private:
  uint8_t i2c_addr;
  boolean _framInitialised;
};

/**************************************************************************/
/*!
    @brief  Writes any types of values at the specific FRAM address.
    		This code is used to write any data type (int, float, double, string, char, etc.) to an I2C FRAM
    		In the normal write anything the framaddress is incrimented after the writing of each byte.
    		The Wire library does this behind the scenes.
    		Maybe maximum size of values to write cannot exceed 32 bytes (max i2c bus buffer size).
    
    @params[in] i2cAddr
                The I2C address of the FRAM memory chip (1010+A2+A1+A0)
    @params[in] framAddr
                The 16-bit address to write to in FRAM memory
    @params[in] i2cAddr
                The 8-bit value to write at framAddr
*/
/**************************************************************************/

template <class T> uint16_t FRAM_I2C::write(uint16_t framAddr, const T& value)
{
const uint8_t* p = (const byte*)(const void*)&value;
uint16_t i;
Wire.beginTransmission(i2c_addr);
Wire.write((uint16_t)(framAddr >> 8)); // MSB
Wire.write((uint16_t)(framAddr & 0xFF)); // LSB
for (i = 0; i < sizeof(value); i++) Wire.write(*p++);
Wire.endTransmission();
return i;
}

template <class T> uint16_t FRAM_I2C::read(uint16_t framAddr, T& value)
{
uint8_t* p = (byte*)(void*)&value;
uint16_t i;
Wire.beginTransmission(i2c_addr);
Wire.write((uint16_t)(framAddr >> 8)); // MSB
Wire.write((uint16_t)(framAddr & 0xFF)); // LSB
Wire.endTransmission();
Wire.requestFrom(i2c_addr,sizeof(value));
for (i = 0; i < sizeof(value); i++)
if(Wire.available())
*p++ = Wire.read();
return i;
}

#endif

Here is an example:

#include <Wire.h>
#include "Adafruit_FRAM_I2C.h"

struct config
{
long targetLat;
long targetLon;
float fNum;
bool first;
int attempts;
} config;

Adafruit_FRAM_I2C fram = Adafruit_FRAM_I2C();

void setup()
{

Serial.begin(9600);

//config some of the variables during to be saved. 
config.targetLat = 4957127;
config.targetLon = 6743421;
config.first = false;
config.attempts = 30;
config.fNum = 2.23;
fram.write(0,config);
delay (30);
}

void loop()
{

//change the variables within the structure to show the read worked
config.targetLat =0;
config.targetLon = 0;
config.first = true;
config.fNum = 0.0;
config.attempts = 0;

fram.read(0,config);
delay(30);

Serial.print("lat = ");
Serial.println(config.targetLat);
Serial.print("lon =");
Serial.println(config.targetLon);

Serial.print("first");
Serial.println(config.first);

Serial.print("float");
Serial.println(config.fNum);

Serial.print("attempts = ");
Serial.println(config.attempts);

delay(1000);
}

josharian, 2016-06-01:
I'm always nervous about "serializing" by copying memory, but it would be helpful to have methods in the main library that read and write from buffers. It is considerably more efficient than writing individual bytes if you have a lot of data to write. We'd want to document the wraparound behavior near the end of the FRAM address space.

> Maybe maximum size of values to write cannot exceed 32 bytes (max i2c bus buffer size).

Would that be 30, since two bytes are used to write the address? Have you tested to confirm that this is in fact a real limit? I've experimented with large buffer reads (up to 1024 bytes) and had no problems; haven't tested large writes yet.

josharian, 2016-06-01:
@ladyada @andydoro if there's interest in buffer reads/writes, I'm happy to send a PR.

josharian, 2016-06-03:
Since there's no feedback yet about a PR...for anyone else who comes across this, here are buffer-oriented read and write functions, for `uint8_t`-sized buffers. They aren't incorporated into the library, so have a hard-coded `0x50` i2c address. They gracefully handle buffers larger than the i2c bus buffer size.

```
uint8_t readn(uint16_t framAddr, uint8_t n, uint8_t *p)
{
  uint8_t i = 0;
  for(i = 0; i < n; ) {
    Wire.beginTransmission(0x50);
    Wire.write(framAddr >> 8);
    Wire.write(framAddr & 0xFF);
    Wire.endTransmission();
    uint8_t req;
    req = n;
    if(n > 32) {
      req = 32;
    }
    Wire.requestFrom(uint8_t(0x50), req);
    uint8_t j;
    for (j = 0; j < 32 && i < n; i++, j++) {
      if(!Wire.available()) {
        return i;
      }
      *p++ = Wire.read();
    }
    framAddr += j;
  }
  return i;
}

void writen(uint16_t framAddr, uint8_t n, uint8_t *p)
{
  uint8_t i = 0;
  for(i = 0; i < n; ) {
    Wire.beginTransmission(0x50);
    Wire.write(framAddr >> 8);
    Wire.write(framAddr & 0xFF);
    uint8_t j;
    for (j = 0; j < 30 && i < n; i++, j++) {
      Wire.write(*p++);
    }
    Wire.endTransmission();
    framAddr += j;
  }
}
```

bcraigie, 2016-11-26:
The code contributed by @gabrydiamonds at the top of this page looks very useful, but it won't compile (presumably because it wasn't properly pasted as code.  There is a `*` missing in the comments - `/!` should be `/*!` for example.  There are `#` symbols missing at the ifndef  and endif.  There are `<` and `>` symbols missing in the template lines.

I fixed most of them but can't figure this one out as I've never come across templates before (I added the < and >):

`template <uint16_t> write(uint16_t framAddr, const T& value);`

- It fails at the T& (error: 'T' does not name a type).

Does anyone have a working version of this code?

Thanks!

ladyada, 2021-07-01:
hiya we are adding readObject and writeObject for class read/write help. theres a String exmaple as well

closed 2021-07-01 by ladyada