adafruit/Adafruit_FRAM_I2C #11: How to store large numbers

EliasA97, 2019-10-07:
Hi, first of all I must say that the library works just fine, but I want to store numbers like 800.000 or 1.000.000. How can this be done?

ismailtan, 2019-10-09:
Just got two of these modules. I am also looking for  a way to store long's, or at least an int.

EliasA97, 2019-10-09:
I found this: http://mario.mtechcreations.com/programing/my-adafruit-fram-i2c-plus-library/
It might help us both!!

ismailtan, 2019-10-09:
That seems promising, thanks! I will have a better look at it later this week/begin next week.

I have it working now like this without a library. With the library of mtechcreations the code would probably look a lot cleaner.

`#include <Wire.h>
#include "Adafruit_FRAM_I2C.h"

Adafruit_FRAM_I2C fram     = Adafruit_FRAM_I2C();
uint16_t          addr = 0;

struct data {
  long admin;
  long callTimeM;
  long calls;
  long buttonPresses;
  long programMode;
  long startTime;
  long endTime;
  long volume;
  long batteryStatus;
  long timeStamp;
};

data config_1 = {
  612345678,  // admin
  300,        // callTimeM
  20,         // calls
  25,         // buttonPresses
  1,          // programMode
  9,          // startTime
  17,         // endTime
  5,          // volume
  100,        // batteryStatus
  0           // timeStamp
};

union ui32_to_ui8 {
  uint32_t ui32;
  uint8_t ui8[4];
};

uint32_t adminReceived;
uint32_t callTimeMReceived;
uint32_t callsReceived;
uint32_t buttonPressesReceived;
uint32_t programModeReceived;
uint32_t startTimeReceived;
uint32_t endTimeReceived;
uint32_t volumeReceived;
uint32_t batteryStatusReceived;
uint32_t timeStampReceived;

uint32_t adminMemLocation         = 0;
uint32_t callTimeMMemLocation     = 8;
uint32_t callsMemLocation         = 16;
uint32_t buttonPressesMemLocation = 24;
uint32_t programModeMemLocation   = 32;
uint32_t startTimeMemLocation     = 40;
uint32_t endTimeMemLocation       = 48;
uint32_t volumeMemLocation        = 56;
uint32_t batteryStatusMemLocation = 64;
uint32_t timeStampMemLocation     = 72;

ui32_to_ui8 admin;
ui32_to_ui8 callTimeM;
ui32_to_ui8 calls;
ui32_to_ui8 buttonPresses;
ui32_to_ui8 programMode;
ui32_to_ui8 startTime;
ui32_to_ui8 endTime;
ui32_to_ui8 volume;
ui32_to_ui8 batteryStatus;
ui32_to_ui8 timeStamp;

void setup(void) {
  Serial.begin(57600);
  fram.begin();

  // ----- uint32_t => uint8_t array ------------------------
  admin.ui32          = config_1.admin;
  callTimeM.ui32      = config_1.callTimeM;
  calls.ui32          = config_1.calls;
  buttonPresses.ui32  = config_1.buttonPresses;
  programMode.ui32    = config_1.programMode;
  startTime.ui32      = config_1.startTime;
  endTime.ui32        = config_1.endTime;
  volume.ui32         = config_1.volume;
  batteryStatus.ui32  = config_1.batteryStatus;
  timeStamp.ui32      = config_1.timeStamp;

  // ----- write data to memory ------------------------
  for (int i = 0 ; i < 4; ++i) {
    fram.write8(adminMemLocation + i, admin.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(callTimeMMemLocation + i, callTimeM.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(callsMemLocation + i, calls.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(buttonPressesMemLocation + i, buttonPresses.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(programModeMemLocation + i, programMode.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(startTimeMemLocation + i, startTime.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(endTimeMemLocation + i, endTime.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(volumeMemLocation + i, volume.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(batteryStatusMemLocation + i, batteryStatus.ui8[i]);
  }
  for (int i = 0; i < 4; ++i) {
    fram.write8(timeStampMemLocation + i, timeStamp.ui8[i]);
  }

  // ----- uint8_t array from memory => uint32_t ------------------------
  for (int i = 0; i < 4; ++i) {
    admin.ui8[i] = fram.read8(adminMemLocation + i);
  }
  adminReceived = admin.ui32;

  for (int i = 0; i < 4; ++i) {
    callTimeM.ui8[i] = fram.read8(callTimeMMemLocation + i);
  }
  callTimeMReceived = callTimeM.ui32;

  for (int i = 0; i < 4; ++i) {
    calls.ui8[i] = fram.read8(callsMemLocation + i);
  }
  callsReceived = calls.ui32;

  for (int i = 0; i < 4; ++i) {
    buttonPresses.ui8[i] = fram.read8(buttonPressesMemLocation + i);
  }
  buttonPressesReceived = buttonPresses.ui32;

  for (int i = 0; i < 4; ++i) {
    programMode.ui8[i] = fram.read8(programModeMemLocation + i);
  }
  programModeReceived = programMode.ui32;

  for (int i = 0; i < 4; ++i) {
    startTime.ui8[i] = fram.read8(startTimeMemLocation + i);
  }
  startTimeReceived = startTime.ui32;

  for (int i = 0; i < 4; ++i) {
    endTime.ui8[i] = fram.read8(endTimeMemLocation + i);
  }
  endTimeReceived = endTime.ui32;

  for (int i = 0; i < 4; ++i) {
    volume.ui8[i] = fram.read8(volumeMemLocation + i);
  }
  volumeReceived = volume.ui32;

  for (int i = 0; i < 4; ++i) {
    batteryStatus.ui8[i] = fram.read8(batteryStatusMemLocation + i);
  }
  batteryStatusReceived = batteryStatus.ui32;

  for (int i = 0; i < 4; ++i) {
    timeStamp.ui8[i] = fram.read8(timeStampMemLocation + i);
  }
  timeStampReceived = timeStamp.ui32;

  Serial.print("Admin phone_nr      : 0");
  Serial.println(adminReceived);
  Serial.print("Minutes called      : ");
  Serial.println(callTimeMReceived);
  Serial.print("Received calls      : ");
  Serial.println(callsReceived);
  Serial.print("Button presses      : ");
  Serial.println(buttonPressesReceived);

  Serial.print("Program mode        : ");
  if (programModeReceived == 1) {
    Serial.println("always on ");
  }
  else {
    Serial.println("not set");
  }

  Serial.print("Time start of day   : ");
  Serial.println(startTimeReceived);
  Serial.print("Time end of day     : ");
  Serial.println(endTimeReceived);
  Serial.print("Audio volume        : ");
  Serial.println(volumeReceived);
  Serial.print("Battery status      : ");
  Serial.println(batteryStatusReceived);
  Serial.print("Last data timestamp : ");
  Serial.println(timeStampReceived);
}

void loop(void) {

}`

ladyada, 2021-07-01:
hiya we are adding readObject and writeObject for class read/write help. theres a String exmaple as well

closed 2021-07-01 by ladyada