adafruit/Adafruit_FRAM_I2C #12: Testing on the Jetson Nano/TX2 Failed

S4WRXTTCS, 2019-12-04:
I've been testing quite a few Adafruit I2C Sensors/Devices on the Jetson Nano using Python, and for the most part they seem to work.  Sometimes with weird issues, but nothing that likely isn't easily solvable.

This is what I used as a starting point.
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano?view=all

But, when I tried the I2C FRAM it didn't work in a rather bizarre way. 

It came back with "No I2C device at address 7c"

That's rather odd because the default address for this device is 0x50. I also don't see an address of 0x7c anywhere in any of the code. So I don't understand where it's getting that address from. I tried other examples with the devices unplugged, and they all gave the proper addresses when reporting that same error.  Where the address matched the expected default address of the device 

One other abnormality is the FRAM I2C CircuitPython example has board.SDA before board.SCL, but all the other examples I've tried has them the other way around.   
https://learn.adafruit.com/adafruit-i2c-fram-breakout/python-circuitpython

caternuson, 2019-12-05:
The weird address comes from the odd process for querying the device ID. See data sheet page 10:
https://cdn-learn.adafruit.com/assets/assets/000/043/904/original/MB85RC256V-DS501-00017-3v0-E.pdf?1500009796
and the `getDeviceID()` method in this library.

This is the Arduino library. Are you having issues with the CircuitPython library?

S4WRXTTCS, 2019-12-05:
I'm using the CircuitPython on regular Python so I most definitely posted in the wrong spot. 

Oops. I'll close this, and find the right spot to post this.

Thanks for your help.

closed 2019-12-05 by S4WRXTTCS

caternuson, 2019-12-05:
Cool. No worries. Thanks for responding and closing. Here's the CircuitPython library repo:
https://github.com/adafruit/Adafruit_CircuitPython_FRAM