adafruit/Adafruit_FRAM_I2C #22: Latest version (2.0.0) of the library: begin() fails with the FRAM breakout board (Product ID: 1895)

dlveach, 2022-04-08:
- Arduino board:  Nano 33 BLE

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.15

- List the steps to reproduce the problem below 

Brand new (just purchased) Adafruit FRAM I2C breakout board.  Product ID 1895.
Install latest version of the Adafruit_FRAM_I2C library (2.0.0)
Using default I2C address 0x50  (confirmed with I2C_Scanner sketch)
Code fails in Adafruit_FRAM_I2C.cpp call to begin():
`
bool Adafruit_FRAM_I2C::begin(uint8_t addr, TwoWire *theWire) {
  Adafruit_EEPROM_I2C::begin(addr, theWire);

  // the MB85 has a secondary address too!
  i2c_dev2 = new Adafruit_I2CDevice(MB85RC_SECONDARY_ADDRESS, theWire);
  if (!i2c_dev2->begin()) {
    return false;
  }

     < ... >
`

Take note of:
`
#define MB85RC_DEFAULT_ADDRESS                                                 \
  (0x50) ///<* 1010 + A2 + A1 + A0 = 0x50 default */
#define MB85RC_SECONDARY_ADDRESS                                               \
  (0x7C) ///< secondary ID for manufacture id info
`
I2C_Scanner sketch detected the 0x50 default address however did NOT detect the secondary address 0x7C.  Does not appear to be present in this particular FRAM chip.

After downgrading the library to ver 1.3.3 that does not have the attempt to connect to the secondary I2C address everything works fine and the begin() fn returns true as expected.

ladyada, 2022-04-08:
hi can you format the issue so we can tell where the code starts and begins

dlveach, 2022-04-08:
Sorry, I broke the markup.   Should be clear now?   Lemme know if you need more.

caternuson, 2022-04-08:
Can't recreate this easily. Connected an I2C FRAM breakout ([PID 1895](https://www.adafruit.com/product/1895)) to an Itsy nRF52840 ([PID 4481](https://www.adafruit.com/product/4481)), closest I have to a Nano 33 BLE.

The I2C scanner sketch (from here) reports both addresses:
[image]

The [MB85RC256V example](https://github.com/adafruit/Adafruit_FRAM_I2C/blob/master/examples/MB85RC256V/MB85RC256V.ino) from this library also runs as expected:
[image]

@dlveach Suggest posting in the forums with photos of your setup so we can look for any potential hardware issues:
https://forums.adafruit.com/

dlveach, 2022-04-08:
I'll do that.   I have this all on a breadboard and I do have other I2C devices connected.   You made me think, I should try isolating the FRAM breakout alone without any other I2C devices and running Scanner before I do that.  Just in case there's some kind of interaction going on causing the secondary address not found.   Thank you.

dlveach, 2022-04-08:
Posted.  [https://forums.adafruit.com/viewtopic.php?f=22&t=190130](url)

dlveach, 2022-05-04:
Just following up to add that after discussion in the above post on the forums with "adafruit_support_mike", including some O-scope captures to confirm I2C comm is working, we confirmed the Adafruit board is just not responding to this secondary address check added in the 1.3.3 -> 2.0.0 library change.   Unless this change is rolled back it looks like I'm stuck on version 1.3.3 to continue using it.

jacob-newormics, 2023-03-03:
This same exact thing happened to me in my setup with the same product at the `i2c_dev` conditional even though it was showing up in a generic I2C scan. Reverting to 1.1.3 solved it.

Happy to try troubleshooting with you on this @caternuson